Meebleeps/MeeBleeps-Freaq-FM-Synth #3: Does it work on a UNO?

mayasfinest, 2022-06-30:
Hi,
excuse my little knowledge.
I also have problems with the size of the code but feel unable to figure out what to change to make it work on my side.
I used the standard Arduino Software trying to upload your code on a Nano compatible, no way.
But then I tried a UNO I had also lying around, the upload succeeded, just telling me "Low memory available, stability problems may occur."

Will it be possible to run the whole synth on a UNO instead and is there any change in pinout or behaviour...?
Or is it just fully compatible...?

Thanks for this gread project!
Cheers
Jörg

Meebleeps, 2022-07-01:
Hi Jörg,

You _can_ run it on the Uno, but the design uses the Nano's extra 2 Analog inputs, so you will not have control over the FM ratio and depth, so it won't be very good!  In theory it _could_ be redesigned but would take a lot of coding to assign secondary functions to other knobs and wouldn't be as fun to use.

To make it fit in the Nano, try this:

Open the file **MutantFMSynthOptions.h**

find the line:
`//#define COMPILE_SMALLER_BINARY`

now remove the commenting symbol **//** from the start so it looks like this:

`#define COMPILE_SMALLER_BINARY`

Then compile it.  It should now easily fit into the Nano by leaving out one of the wave tables - you will still have plenty to play with and probably won't miss it! :)

good luck!

mayasfinest, 2022-07-01:
Great, thanks!

The upload succeeded but I got a warning:

"Sketch uses 28938 bytes (94%) of program storage space. Maximum is 30720 bytes.
Global variables use 1778 bytes (86%) of dynamic memory, leaving 270 bytes for local variables. Maximum is 2048 bytes.
Low memory available, stability problems may occur."

I'll build the hardware now and see how it rocks, I'm excitied!

Meebleeps, 2022-07-01:
awesome, glad it worked.  yes that warning is normal - the code is designed not to use much local variable memory so it hasn't caused any issues (yet! 😁).

Good luck with the build - I'd love it if you shared video or pics when you've got it up and running! 😁 if you have any questions during the build, let me know and I'll try to help.

mayasfinest, 2022-07-01:
...is the Midi-Out fully implanted and working?
Does it  send all notes it produces out to channel1?
Can I change the channel somewhere in the code?

Thnx

mayasfinest, 2022-07-02:
Thanks mate for the extensive answer!

One last thing, on the schematic it says "0.1mF" at two posititons. I'd never seen this before.
At first I thought it was a type, meaning either 0.1nF or 0.1uF (micro), then realising it is a polarized cap.
Could you help me out, what do I need here...?

Thanks!

Meebleeps, 2022-07-03:
Hi, no worries. I used 100uF caps, which technically are 0.1mF but don’t know why fritzing displayed it like that!   the values were a bit trial and error - this was my second electronics project ever so self/YouTube-learning as I went!

mayasfinest, 2022-07-05:
This is what I did to your Synth!
[image]

[image]

Meebleeps, 2022-07-06:
I love it and I love the DOTT theme.  Awesome! 😁

Congratulations from Western Australia! 😊