Meebleeps/MeeBleeps-Freaq-FM-Synth #4: Permanant hiss in output

mayasfinest, 2022-07-06:
Me again,
i am getting a rather noticable but quite annoying med high frequency in my output (f#'').
When I decouple it with values bigger then 1uF it gets completely filtered out but the overall tone is too muffled.
Do you have any further ideas? Can you reproduce it? Where is the hiss produced....

PS: Doesn't make any difference if I run it from PSU or battery

Cheers,
Jörg

mayasfinest, 2022-07-06:
OK, i figured it out, in case anyone else also has this problem here's the solution:

I don't know why but the source of the hiss is the matrix display and to get rid of it i simply removed (!) the GND cable from the matrix-board, leaving just the 100uF between VCC+GND. The display is still fully functioning..

sergioska, 2023-05-15:
uhm ... same problem. I tried @mayasfinest solution but nothing ... it doesn't work, hiss doesn't go away.
I listen this background noise also in some videos on youtube, so I think that someone else has this problem
Does anyone find a solution?

sergioska, 2023-05-22:
Thanks @Meebleeps i'll try your suggestions!