Meebleeps/MeeBleeps-Freaq-FM-Synth #5: Testing the beast

spherical-sound-society, 2022-07-08:
Im building one, just in the test phase.

I have the arduino nano correctly flashed and connected just the audio output and the 8x8led matrix.

The switches are congig as "normally open"

When I power it I get no leds on and just a faint background humm in the audio.

Should I simulate the press of any key to get it reacting? Im not sure if the led matrix is compatible and I dont feel like building the whole enchilada to later discover that the led matrix is not suitable.

This one uses the max7219 chip

If I manage to get it working I would love to try to update it for using a bluepill (wich is mozzi compatible too) so I can try to get way more voices and keep developing it.

But first lets go little by little!
Thanks!!

spherical-sound-society, 2022-07-08:
I have it connected like this, by the way:

clk>sck
miso>din
cs led>cs
gnd>gnd
vcc>vcc

Meebleeps, 2022-07-09:
Hi,

I am remote and don’t have decent internet access, so sorry I can’t help more. But I can confirm the led matrix uses a MAX7219 driver. The D_in pin on the Matrix should be connected to the arduino’s MOSI pin (Master Out Slave In).

spherical-sound-society, 2022-07-10:
Hi!
Yep, I had it right connected but I still get no sound or leds working.

No hurry about it, just reply when you come back from holydays :)

It is any thing that I can do for it to start working? Maybe it needs to press start or something?

Meebleeps, 2022-07-11:
Hi, when you power it on, the display should flash with a Meebleeps logo, so if that doesn’t happen, you have some problem with power, arduino, led matrix, or the connections between these. 

If you get the logo but no sound, then it may be a problem with the audio output circuit or the button configuration. and yes maybe bash start a few times and see what happens 😊

spherical-sound-society, 2022-07-11:
Great. That´s exactly what I needed to know. I will try again.

Thanks!

spherical-sound-society, 2022-07-11:
Ok, it turns out that the cable was busted. I have tried with a dupons of mine instead the ones coming with the matrix and I have it working!! :D
Thank you so much!

closed 2022-07-11 by spherical-sound-society

Meebleeps, 2022-10-11:
Great to hear you got it working! 😊