Meebleeps/MeeBleeps-Freaq-FM-Synth #15: Can't compile on Arduino IDE

cmdilger, 2025-01-28:
When I try to compile, these errors appear, how can I fix this?

C:\Users\cmdil\AppData\Local\arduino\sketches\6F27CEE6481A32DDBA98EDF912363EBC\sketch\avSequencer.cpp.o (symbol from plugin): In function `MozziPrivate::output_buffer':
(.text+0x0): multiple definition of `MozziPrivate::output_buffer'
C:\Users\cmdil\AppData\Local\arduino\sketches\6F27CEE6481A32DDBA98EDF912363EBC\sketch\MutantFMSynth.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here
C:\Users\cmdil\AppData\Local\arduino\sketches\6F27CEE6481A32DDBA98EDF912363EBC\sketch\avSequencer.cpp.o (symbol from plugin): In function `MozziPrivate::output_buffer':
(.text+0x0): multiple definition of `MozziPrivate::MozziRandPrivate::x'
C:\Users\cmdil\AppData\Local\arduino\sketches\6F27CEE6481A32DDBA98EDF912363EBC\sketch\MutantFMSynth.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here
C:\Users\cmdil\AppData\Local\arduino\sketches\6F27CEE6481A32DDBA98EDF912363EBC\sketch\avSequencer.cpp.o (symbol from plugin): In function `MozziPrivate::output_buffer':
(.text+0x0): multiple definition of `MozziPrivate::MozziRandPrivate::y'
C:\Users\cmdil\AppData\Local\arduino\sketches\6F27CEE6481A32DDBA98EDF912363EBC\sketch\MutantFMSynth.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here
C:\Users\cmdil\AppData\Local\arduino\sketches\6F27CEE6481A32DDBA98EDF912363EBC\sketch\avSequencer.cpp.o (symbol from plugin): In function `MozziPrivate::output_buffer':
(.text+0x0): multiple definition of `MozziPrivate::MozziRandPrivate::z'
C:\Users\cmdil\AppData\Local\arduino\sketches\6F27CEE6481A32DDBA98EDF912363EBC\sketch\MutantFMSynth.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here
C:\Users\cmdil\AppData\Local\arduino\sketches\6F27CEE6481A32DDBA98EDF912363EBC\sketch\avSequencer.cpp.o (symbol from plugin): In function `MozziPrivate::output_buffer':
(.text+0x0): multiple definition of `MozziPrivate::adcPinToChannelNum(unsigned char)'
C:\Users\cmdil\AppData\Local\arduino\sketches\6F27CEE6481A32DDBA98EDF912363EBC\sketch\MutantFMSynth.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here
C:\Users\cmdil\AppData\Local\arduino\sketches\6F27CEE6481A32DDBA98EDF912363EBC\sketch\avSequencer.cpp.o (symbol from plugin): In function `MozziPrivate::output_buffer':
(.text+0x0): multiple definition of `MozziPrivate::adcStartConversion(unsigned char)'
C:\Users\cmdil\AppData\Local\arduino\sketches\6F27CEE6481A32DDBA98EDF912363EBC\sketch\MutantFMSynth.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here
C:\Users\cmdil\AppData\Local\arduino\sketches\6F27CEE6481A32DDBA98EDF912363EBC\sketch\avSequencer.cpp.o (symbol from plugin): In function `MozziPrivate::output_buffer':
(.text+0x0): multiple definition of `MozziPrivate::setupFastAnalogRead(signed char)'
C:\Users\cmdil\AppData\Local\arduino\sketches\6F27CEE6481A32DDBA98EDF912363EBC\sketch\MutantFMSynth.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here
C:\Users\cmdil\AppData\Local\arduino\sketches\6F27CEE6481A32DDBA98EDF912363EBC\sketch\avSequencer.cpp.o (symbol from plugin): In function `MozziPrivate::output_buffer':
(.text+0x0): multiple definition of `MozziPrivate::setupMozziADC(signed char)'
C:\Users\cmdil\AppData\Local\arduino\sketches\6F27CEE6481A32DDBA98EDF912363EBC\sketch\MutantFMSynth.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here
C:\Users\cmdil\AppData\Local\arduino\sketches\6F27CEE6481A32DDBA98EDF912363EBC\sketch\avSequencer.cpp.o (symbol from plugin): In function `MozziPrivate::output_buffer':
(.text+0x0): multiple definition of `MozziPrivate::stopMozzi()'
C:\Users\cmdil\AppData\Local\arduino\sketches\6F27CEE6481A32DDBA98EDF912363EBC\sketch\MutantFMSynth.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here
C:\Users\cmdil\AppData\Local\arduino\sketches\6F27CEE6481A32DDBA98EDF912363EBC\sketch\avSequencer.cpp.o (symbol from plugin): In function `MozziPrivate::output_buffer':
(.text+0x0): multiple definition of `MozziPrivate::MozziRandPrivate::autoSeed()'
C:\Users\cmdil\AppData\Local\arduino\sketches\6F27CEE6481A32DDBA98EDF912363EBC\sketch\MutantFMSynth.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here
C:\Users\cmdil\AppData\Local\arduino\sketches\6F27CEE6481A32DDBA98EDF912363EBC\sketch\avSequencer.cpp.o (symbol from plugin): In function `MozziPrivate::output_buffer':
(.text+0x0): multiple definition of `MozziPrivate::audioTicks()'
C:\Users\cmdil\AppData\Local\arduino\sketches\6F27CEE6481A32DDBA98EDF912363EBC\sketch\MutantFMSynth.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here
C:\Users\cmdil\AppData\Local\arduino\sketches\6F27CEE6481A32DDBA98EDF912363EBC\sketch\avSequencer.cpp.o (symbol from plugin): In function `MozziPrivate::output_buffer':
(.text+0x0): multiple definition of `MozziPrivate::mozziMicros()'
C:\Users\cmdil\AppData\Local\arduino\sketches\6F27CEE6481A32DDBA98EDF912363EBC\sketch\MutantFMSynth.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here
C:\Users\cmdil\AppData\Local\arduino\sketches\6F27CEE6481A32DDBA98EDF912363EBC\sketch\avSequencer.cpp.o (symbol from plugin): In function `MozziPrivate::output_buffer':
(.text+0x0): multiple definition of `MozziPrivate::startMozzi(int)'
C:\Users\cmdil\AppData\Local\arduino\sketches\6F27CEE6481A32DDBA98EDF912363EBC\sketch\MutantFMSynth.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here
C:\Users\cmdil\AppData\Local\arduino\sketches\6F27CEE6481A32DDBA98EDF912363EBC\sketch\avSequencer.cpp.o (symbol from plugin): In function `MozziPrivate::output_buffer':
(.text+0x0): multiple definition of `mozziMicros()'
C:\Users\cmdil\AppData\Local\arduino\sketches\6F27CEE6481A32DDBA98EDF912363EBC\sketch\MutantFMSynth.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here
C:\Users\cmdil\AppData\Local\arduino\sketches\6F27CEE6481A32DDBA98EDF912363EBC\sketch\avSequencer.cpp.o (symbol from plugin): In function `MozziPrivate::output_buffer':
(.text+0x0): multiple definition of `audioTicks()'
C:\Users\cmdil\AppData\Local\arduino\sketches\6F27CEE6481A32DDBA98EDF912363EBC\sketch\MutantFMSynth.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here
C:\Users\cmdil\AppData\Local\arduino\sketches\6F27CEE6481A32DDBA98EDF912363EBC\sketch\avSequencer.cpp.o (symbol from plugin): In function `MozziPrivate::output_buffer':
(.text+0x0): multiple definition of `startMozzi(int)'
C:\Users\cmdil\AppData\Local\arduino\sketches\6F27CEE6481A32DDBA98EDF912363EBC\sketch\MutantFMSynth.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here
C:\Users\cmdil\AppData\Local\arduino\sketches\6F27CEE6481A32DDBA98EDF912363EBC\sketch\avSequencer.cpp.o (symbol from plugin): In function `MozziPrivate::output_buffer':
(.text+0x0): multiple definition of `stopMozzi()'
C:\Users\cmdil\AppData\Local\arduino\sketches\6F27CEE6481A32DDBA98EDF912363EBC\sketch\MutantFMSynth.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here
C:\Users\cmdil\AppData\Local\arduino\sketches\6F27CEE6481A32DDBA98EDF912363EBC\sketch\avSequencer.cpp.o (symbol from plugin): In function `MozziPrivate::output_buffer':
(.text+0x0): multiple definition of `setupMozziADC(signed char)'
C:\Users\cmdil\AppData\Local\arduino\sketches\6F27CEE6481A32DDBA98EDF912363EBC\sketch\MutantFMSynth.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here
C:\Users\cmdil\AppData\Local\arduino\sketches\6F27CEE6481A32DDBA98EDF912363EBC\sketch\avSequencer.cpp.o (symbol from plugin): In function `MozziPrivate::output_buffer':
(.text+0x0): multiple definition of `setupFastAnalogRead(signed char)'
C:\Users\cmdil\AppData\Local\arduino\sketches\6F27CEE6481A32DDBA98EDF912363EBC\sketch\MutantFMSynth.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here
C:\Users\cmdil\AppData\Local\arduino\sketches\6F27CEE6481A32DDBA98EDF912363EBC\sketch\avSequencer.cpp.o (symbol from plugin): In function `MozziPrivate::output_buffer':
(.text+0x0): multiple definition of `adcPinToChannelNum(unsigned char)'
C:\Users\cmdil\AppData\Local\arduino\sketches\6F27CEE6481A32DDBA98EDF912363EBC\sketch\MutantFMSynth.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here
C:\Users\cmdil\AppData\Local\arduino\sketches\6F27CEE6481A32DDBA98EDF912363EBC\sketch\avSequencer.cpp.o (symbol from plugin): In function `MozziPrivate::output_buffer':
(.text+0x0): multiple definition of `audioOutput(MonoOutput)'
C:\Users\cmdil\AppData\Local\arduino\sketches\6F27CEE6481A32DDBA98EDF912363EBC\sketch\MutantFMSynth.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here
[... cut ...]
C:\Users\cmdil\AppData\Local\arduino\sketches\6F27CEE6481A32DDBA98EDF912363EBC\sketch\MutantFMSynth.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here
C:\Users\cmdil\AppData\Local\arduino\sketches\6F27CEE6481A32DDBA98EDF912363EBC\sketch\avSourceFM.cpp.o (symbol from plugin): In function `MutatingFM::setGain(unsigned char)':
(.text+0x0): multiple definition of `audioHook()'
C:\Users\cmdil\AppData\Local\arduino\sketches\6F27CEE6481A32DDBA98EDF912363EBC\sketch\MutantFMSynth.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here
collect2.exe: error: ld returned 1 exit status

exit status 1

Compilation error: exit status 1

joniuz, 2025-01-28:
I have the same problem here, it's seems that the Mozzi library used to compile the repository code was 1.3 and currently the version is 2.0.

There is still a 1.x branch called “Mozzi_1” in the git repository (As the Mozzi documentation said).  I tried to compile with that version and I got the same "multiple definition of" errors. 

I'm using VSCode and PlatformIO.

cmdilger, 2025-01-28:
I installed VSCode and PlatformIO, but several errors still appear, I don't know how to fix them...
Could you help me?
How should I change it to compile and burn it to Arduino?

cmdilger, 2025-01-29:
Now only this error remains, does anyone know how to fix it?

	"message": "'uint16_t mozziAnalogRead(uint8_t)' is deprecated: This use of mozziAnalogRead() is not portable. Refer to the API documentation for suggested alternatives. [-Wdeprecated-declarations]"

closed 2025-01-29 by cmdilger

joniuz, 2025-01-29:
VSCode also didn't work to me. Finally I managed to compile the code, but using Arduino IDE.
You must copy the src folder from the repository to another folder and rename it to MutantFMSynth

Open the Arduino IDE and install the Mozzi library using the Library Manager, I choose the version 1.1.2.

Then open the file MutantFMSynth.ino that is inside the folder and edit the file MutantFMSynthOptions.h uncommenting the line  21
//#define COMPILE_SMALLER_BINARY

Also verify the lines 9 and 12 that configure your switch type, in my case they are normally open

#undef SWITCH_TYPE_NORMALLY_CLOSED

Now I'm dealing with the pots and switches, it seems the schematic is not correct.